nasa/cFE #2174: ES CRC type enumerations do not belong in cfe_mission_cfg.h

jphickey, 2022-10-20:
**Is your feature request related to a problem? Please describe.**
The mission config header file contains the following definitions:

https://github.com/nasa/cFE/blob/be88a07313ff3b579c8abaef4b4a9085e74aabdf/cmake/sample_defs/sample_mission_cfg.h#L261-L268

These enumeration values are really part of the API to the `CFE_ES_CalculateCRC()` function, as they define the values for the 4th parameter.   

They do not belong in "mission_cfg.h" because are not values that appear in any telemetry, nor in any command, nor would they need to be configured differently as there is no chance of conflict with anything else.

A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
Make a proper enumeration type in the in the `cfe_es_api_typedefs.h` file, and use this for the 4th parameter of the `CFE_ES_CalculateCRC()` function.

**Additional context**
The "DEFAULT" value is the only thing that (theoretically) does belong here:

https://github.com/nasa/cFE/blob/be88a07313ff3b579c8abaef4b4a9085e74aabdf/cmake/sample_defs/sample_mission_cfg.h#L281

However, this is just a placeholder anyway because only the CRC16 algorithm was ever implemented.  Unless the user also has a patched version of ES, no user can set this to anything other than CRC16 or else the system will not function correctly.  Because this is somewhat of a "dead" option, and a case could be made to remove it entirely.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2022-10-20:
This was noted because when using EDS, all the mission-scope configurable items (i.e. everything in cfe_mission_cfg.h) need to be defined in an EDS so they can be referenced from EDS files and scripts.  

This enumeration stood out because it was an extraneous item defined in the C header but not actually used in the CMD/TLM interface.

thnkslprpt, 2022-10-31:
I had a crack at this - including removal of the default option.

commit d6662149 (2022-12-20): Merge pull request #2192 from thnkslprpt/fix-2174-move-crc-types-into-enum

Fix #2174, Move CRC types and convert to enum
commit ed4722b2 (2022-12-20): Fix #2174, Move CRC types and convert to enum
PR #2192 (2022-12-20): Fix #2174, Move CRC types and convert to enum
**Checklist**
* [x] I reviewed the [Contributing Guide](https://github.com/nasa/osal/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Fixes #2174 
The macro definitions of the CRC types (8, 16, and 32-bit CRC polynomials) have been moved to `cfe_es_api_typedefs.h` and turned into a typedef'd `enum`.
Th [... cut]
closed 2022-12-20 by dzbaker