nasa/cFE #2252: cFE_SB branch coverage not 100%

dmknutsen, 2023-03-09:
**Is your feature request related to a problem? Please describe.**
Currently CFE_SB is not seeing full code coverage during the unit tests. There are two conditions that seem to be causing this:

1. In various functions there are checks for - PendingEventId != 0 and Status = CFE_SUCCESS. However, both of these variables are being changed together, such that conditions are mutually exclusive. The end results is that there is no path for PendingEvent to still be non-0 if the status does not equal CFE_SUCCESS.

Example:
    if (Status == CFE_SUCCESS)
    {
        CFE_SB_PipeDescSetFree(PipeDscPtr);
        --CFE_SB_Global.StatTlmMsg.Payload.PipesInUse;
    }
    else if (PendingEventID != 0)
    {
        CFE_SB_Global.HKTlmMsg.Payload.CreatePipeErrorCounter++;
    }

2. Another issue that is occurring is in all usages of switch statements.  Without a default case capturing an unknown event ID, SB assumes that the case will always fall within the defined EIDs - which causes an issue with gcov coverage statistics.

Example:

        switch (PendingEventID)
        {
            case CFE_SB_DEL_PIPE_ERR1_EID:
                CFE_EVS_SendEventWithAppID(CFE_SB_DEL_PIPE_ERR1_EID, CFE_EVS_EventType_ERROR, CFE_SB_Global.AppId,
                                           "Pipe Delete Error:Bad Argument,PipedId %ld,Requestor %s",
                                           CFE_RESOURCEID_TO_ULONG(PipeId), FullName);
                break;
            case CFE_SB_DEL_PIPE_ERR2_EID:
                CFE_EVS_SendEventWithAppID(CFE_SB_DEL_PIPE_ERR2_EID, CFE_EVS_EventType_ERROR, CFE_SB_Global.AppId,
                                           "Pipe Delete Error:Caller(%s) is not the owner of pipe %ld", FullName,
                                           CFE_RESOURCEID_TO_ULONG(PipeId));
                break;
        }

**Describe the solution you'd like**
I believe these were already adjudicated as part of the coverage analysis.  However, they were pointed out by JSC SA and it is likely the non100% branch coverage will keep coming up such that it would be worthwhile to revisit/fix or document why we think the branch coverage meets requirements.  Further, we may want to audit our branch coverage for the other core apps as well.

**Requester Info**
Dan Knutsen
NASA GSFC

dmknutsen, 2023-03-09:
It turns out that this info used to be tracked within the cFE VDD.  It looks like we dropped the info in the latest release and will need to output a version B VDD for Draco-rc4

chillfig, 2023-03-09:
> It turns out that this info used to be tracked within the cFE VDD. It looks like we dropped the info in the latest release and will need to output a version B VDD for Draco-rc4

An updated VDD for Draco-rc4 including dispositions about missing line/function/branch code coverage can be found here: https://github.com/nasa/cFE/releases/download/draco-rc4/cFE_VDD_draco_rc4_revB.docx

thnkslprpt, 2023-05-02:
@dmknutsen - Would something like this be a valid solution to these issues?

[image]

Even though it's a hack, this would both improve coverage, and add the missing default cases for all switch blocks, which is preferred by the relevant coding standards anyway. As long as there are clear comments it seems low/no-risk in terms of introducing bugs in the future.

Or is the last case falling through to default, when that's not actually the intended logic, even worse than leaving these lines uncovered?