nasa/cFE #2258: Add generic status code for "invalid request"

jphickey, 2023-03-20:
**Is your feature request related to a problem? Please describe.**
CFE defines several generic status codes, which are not specific to any module.  These can be used by any app/module to indicate common status conditions, for example: 
 - `CFE_SUCCESS`
 - `CFE_STATUS_NOT_IMPLEMENTED`
 - `CFE_STATUS_REQUEST_ALREADY_PENDING`
 - `CFE_STATUS_EXTERNAL_RESOURCE_FAIL`
 
However none of these are really appropriate for a validation failure for a request - e.g. if a value was out of range, or something of that nature.  

**Describe the solution you'd like**
Add the following generic status codes:
- `CFE_STATUS_VALIDATION_FAILURE`: for requests/inputs that fail structural validations; this one can also be useful with TBL validation functions.
- `CFE_STATUS_RANGE_ERROR`: for requests that pass structural validation, but an input value is outside the acceptable range.
- `CFE_STATUS_INCORRECT_STATE`: for requests that are otherwise valid, but the current system state does not permit its acceptance.

**Describe alternatives you've considered**
Each app could define these status codes, but that would be redundant as these are common validation checks that every app should be doing on its inputs, so a common status code is warranted.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

thnkslprpt, 2023-03-20:
Would help with this as well I think: https://github.com/nasa/CF/issues/57

skliper, 2023-03-21:
I like it.  Even better if we could get to one set of status codes for cFS and a common status type provided from OSAL (extended by cFE/apps if needed).  It's a pain to manage 2 to 3 different status code types.  Maybe even consolidate error reporting and construct a default string or whatever... perror like, etc.

commit 10b63dcd (2023-03-30): Fix #2258, add more generic status codes

Adds several more generic CFE_STATUS_ codes to be used for common
validation and input processing conditions.
commit 24df9914 (2023-03-30): Merge pull request #2259 from jphickey/fix-2258-add-status-codes

Fix #2258, add more generic status codes
PR #2259 (2023-03-30): Fix #2258, add more generic status codes
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFE/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Adds several more generic CFE_STATUS_ codes to be used for common validation and input processing conditions.

Fixes #2258

**Testing perform [... cut]
closed 2023-03-30 by dzbaker