nasa/cFE #2284: Incorrect include directories when using add_cfe_tables outside app

jphickey, 2023-04-06:
**Is your feature request related to a problem? Please describe.**
If the TGTNAME variable is set when this function is called, the function does not examine the include directories of the app (via the APP_NAME parameter).

As a result, when this function is used in a custom setting to provide a mission-specific table definition, the include directories used will be different than when this function is used from the context of the app receipe itself. 

**Describe the solution you'd like**
If APP_NAME is valid and is an app target name, then the target `INCLUDE_DIRECTORIES` for that app should always be included in the table build.  

Currently when this function is used outside the app build context, it tries to get INCLUDE_DIRECTORIES for a "cpu1" target, which does not exist, and produces an error.

**Additional context**
This include directory problem only occurs when trying to use customized table files, rather than the default/example files included with apps.

**Requester Info**
Joseph Hickey, Vantage Systems, Inc.

jphickey, 2023-04-06:
Actually this may not be an issue in the mainline, I may need to check it further, will postpone this.

jphickey, 2023-04-07:
OK - Did a little more isolation here - there is in fact a problem with apps such as "hs" that have more than one table file.

For the framework examples like sch_lab and sample_app, the recommendation has been to name the table as same as the app, in which case the `add_cfe_tables` will detect this relationship and use the `INCLUDE_DIRECTORIES` from the app target, which works fine.

Problem occurs for apps like HS that have multiple tables, as these cannot all have the same name.

A simple solution would be to use the `<APP_NAME>.<TABLE_NAME>` convention in cmake - this is used elsewhere in CFE/CFS, so its not new, and I confirmed that a dot (.) is valid for CMake target names too, so it should work fine.

skliper, 2023-04-07:
I don't follow the issue with not matching the table name.  I add_cfe_tables for SC outside of the SC build, the RTS's specifically.  Doesn't add_cfe_tables use the APP_NAME to figure out the relationship?  I'm a little behind main, so maybe I missed a change?

skliper, 2023-04-07:
Nevermind, read the issue closer... I get it.

skliper, 2023-04-07:
Actually, I don't get it.  Isn't this just using APP_NAME?  `add_cfe_table(sc rts001.c)` works for me outside of SC.  Why would you need the app name in both the APP_NAME field and the table name?

https://github.com/nasa/cFE/blob/7af467efdc15f788d44d67cb807403b10c4f0c99/cmake/arch_build.cmake#L208-L217

jphickey, 2023-04-12:
Yeah, my bad, I kinda went back and forth a bit - I was seeing something odd in some cases.  I narrowed it down to the feature we added a while back to support "alternate" table files that the user can choose at deployment or target start-up time.  That is, from a cpu-specific `install_custom.cmake` file, you can do something like:

`add_cfe_tables(sample_app alt_sample_tbl.c alt2_sample_tbl.c)`

In this case it will build all of these table files, and put them in the install dir, and the choice of which one to load is made later on either by choosing which one to copy to the target OR somehow telling the app which one to actually load.  

It was this case where these alternate/extra tables files were not getting the correct include paths.

jphickey, 2023-04-12:
Part of (my own) confusion here is that I'm trying to isolate diffs between the EDS build and backport relevant stuff, the original issue I was seeing was in the EDS build, but some of the changes there were actually breaking stuff.  I'm still working on isolating what (if anything) needs to change for the GSFC baseline build here.

jphickey, 2023-04-12:
Note that in the case of SC app, it's probably fine because SC uses the `target_include_directories()` as it should.  But sample_app and to_lab framework apps were still relying on directory-scope `include_directories` here.

commit 67faef08 (2023-04-25): Merge pull request #2299 from jphickey/fix-2284-add_cfe_tbl

Fix #2284, improve add_cfe_tables function
commit efb8551a (2023-04-25): Fix #2284, improve add_cfe_tables function

Improves the functionality and adds additional documentation about
how this function is intended to work.  The improvements add some
flexibility and intelligence and should be backward compatible with
existing use cases.

The add_cfe_app() function now adds an additional interface target that
can be referenced when building tables, and this includes the
directory-scope properties from the original app build.  Therefore,
calls to add_cfe_tables from other (non-app) contexts can get the full
set of include directories.

This also makes the target name  [... cut]
PR #2299 (2023-04-25): Fix #2284, improve add_cfe_tables function
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFE/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
Improves the functionality and adds additional documentation about how this function is intended to work.  The improvements add some flexib [... cut]
closed 2023-04-25 by dzbaker