nasa/cFE #2395: Testcase code should not directly include "cfe_test_msgids.h" (yet)

jphickey, 2023-07-14:
**Describe the bug**
The module-specific header file is a relatively new addition.  But for a user that has overridden the previous all-inclusive `cfe_msgids.h` file, including the test_msgids.h directly will cause conflict with the overrides.

**To Reproduce**
Override (only) `cfe_msgids.h` in a build, and change the msgids in here.  Do not override cfe_test_msgids.h.
Attempt to build cfe_testcase -- the `cfe_test_msgids.h` inclusion will bypass the modified msgids, and get defaults, which will conflict / mismatch.

**Expected behavior**
Should be backward compatible with an existing override of `cfe_msgids.h`

**System observed on:**
Debian

**Reporter Info**
Joseph Hickey, Vantage Systems, Inc.

skliper, 2023-07-28:
Reopening w/ Dan's concurrence, this really should be fixed.

jphickey, 2023-08-01:
@skliper  -- Is there a specific problem with your current msgid file?  It should still find (and use) a file called `cpu1_msgids.h` and use that for `cfe_msgids.h` on cpu1.  Can you confirm the content of "cfe_msgids.h" in your build is still as you expected it to be?

And yes - `cfe_test_msgids.h` (along with all the others) will still be generated.  For now, they should be ignored -- nothing includes them, if not using the default msgids.

I will look again into some method of (possibly) avoiding the confusion of generating a file that is not used.  But other than that potentially-confusing aspect - is something broken here?

skliper, 2023-09-01:
@dzbaker Recommend closing this in favor of #2405 due to the issues this causes per comment on #2404 about breaking builds that don't include `cfe_testcase`.

https://github.com/nasa/cFE/pull/2404#issuecomment-1638859197

skliper, 2023-09-01:
Note that I was able to work around my issue by overriding the cfe_test_msgids.h source from target_defs/arch_build_custom.cmake:
`set(TEST_CFGFILE_SRC_cfe_test_msgids ${MISSION_DEFS}/cpu1_msgids.h)`

The another option would be to create a symbolic link in target_defs from cfe_test_msgids.h to cpu1_msgids.h but that was less desireable.

PR #2399 (2023-07-17): Fix #2395, do not directly use cfe_test_msgids.h
**Checklist (Please check before submitting)**

* [x] I reviewed the [Contributing Guide](https://github.com/nasa/cFE/blob/main/CONTRIBUTING.md).
* [x] I signed and emailed the appropriate [Contributor License Agreement](https://github.com/nasa/cFS/blob/main/CONTRIBUTING.md#contributor-license-agreement-cla) to GSFC-SoftwareRelease@mail.nasa.gov and copied cfs-program@lists.nasa.gov.

**Describe the contribution**
For a user that has customized cfe_msgids.h, this will get conflicting values.  If/when users have migrated to module-specific msgid  [... cut]
commit 31908e2e (2023-07-17): Fix #2395, do not directly use cfe_test_msgids.h

For a user that has customized cfe_msgids.h, this will get conflicting
values.  If/when users have migrated to module-specific msgid files,
this will be OK, but for now this can break things.
commit 75b807bc (2023-07-17): Merge pull request #2399 from jphickey/fix-2395-cfe-test-msgid

Fix #2395, do not directly use cfe_test_msgids.h
commit 1c9f44b7 (2023-07-17): Revert "Fix #2395, do not directly use cfe_test_msgids.h"