nasa/cFE #2426: OS_QueueCreate_Impl():119:OS_QueueCreate Error. errno = 38 (Function not implemented)

AndrewFrank-zxy, 2023-08-17:
**Describe the bug**
Hi
I am learning the cFS framework and trying to run it on my arm-linux system. So I created a new `sample_defs/toolchain-arm-linux.cmake` file, the compilation process is ok, but there is a problem when running on the target board.

**To Reproduce**
Steps to reproduce the behavior:
1. copy `toolchain-arm-cortexa8_neon-linux-gnueabi.cmake` to `toolchain-arm-linux.cmake`
2. redefine `CMAKE_C_COMPILER`/`CMAKE_CXX_COMPILER`/`CMAKE_FIND_ROOT_PATH`
3. `SET(cpu1_SYSTEM arm-linux)`
4. run `build.sh` (`make distclean` `make prep` `make install`)
5. cp `build/exe/*` to the target board
6. launch target board with ssh, go to the `exe/cpu1` path and run `./core-cpu1`
7. See error

```
... ...
1980-012-14:03:20.03594 CFE_ES_CreateObjects: Calling CFE_FS_EarlyInit
OS_QueueCreate_Impl():119:OS_QueueCreate Error. errno = 38 (Function not implemented)
1980-012-14:03:20.13712 CFE_EVS_TaskInit: Call to CFE_SB_CreatePipe Failed:RC=0xCA000005
1980-012-14:03:20.13841 CFE_EVS_TaskMain: Application Init Failed,RC=0xCA000005
1980-012-14:03:20.13929 CFE_ES_ExitApp: CORE Application CFE_EVS Had an Init Error.
1980-012-14:03:20.14006 CFE_ES_ResetCFE: PROCESSOR RESET called from CFE_ES_ResetCFE (Commanded).
CFE_PSP: Exiting cFE with PROCESSOR Reset status.
Aborted
```

**System observed on:**
 - Hardware
 - OS: [Linux 5.15.18]
 - Versions [how to know?]

jphickey, 2023-08-18:
The target system / C library does not have POSIX message queues enabled.

Did you build this arm-linux toolchain yourself or is it something that come with another system?  If its from buildroot (or a similar tool) then make sure you have the POSIX message queue option checked.

AndrewFrank-zxy, 2023-08-19:
The reason is found. I used `buildroot` to build a linux program for the first time. I used qemu-vexpress-defconfig. There is no support for mqueue in the kernel config option. I am currently looking for a solution to this problem.

closed 2023-08-22 by AndrewFrank-zxy