nasa/cFE #2434: -Werror=format-truncation warning when building cFE

dmknutsen, 2023-08-22:
**Is your feature request related to a problem? Please describe.**
We get the following warning when building cFE:

/home/dan/Desktop/test/cert_testbed/cfe/modules/evs/fsw/src/cfe_evs_utils.c:548:67: error: ‘snprintf’ output may be truncated before the last format character [-Werror=format-truncation=]
  548 |    snprintf(PortMessage, sizeof(PortMessage), "%s %u/%u/%s %u: %s", TimeBuffer,
 
Typically we disable this warning as a default build option...however some projects like to enable the warning, and this prevents a successful build.

**Describe the solution you'd like**
Size the buffers appropriately by default such that we do not get the warning.

**Requester Info**
Dan Knutsen
NASA Goddard

skliper, 2023-08-23:
If users want to turn on these warnings and want to squash, they can increase `CFE_EVS_MAX_PORT_MSG_LENGTH` such that it's long enough to avoid any case of truncation.  See discussion in #1097.  

It's undesirable to add extra complexity to open source just to avoid a warning that is disabled by default since the point of using snprintf is to truncate when needed.  This is not a bug.  Recommend closing this issue and leave open source as-is.

Code ref:
https://github.com/nasa/cFE/blob/d018aa6ea3b7b268b977b185cd115701a44c9a17/modules/evs/fsw/src/cfe_evs_utils.c#L548-L551

dmknutsen, 2023-08-23:
I like that approach.

closed 2023-08-23 by dmknutsen