nasa/cFE #2439: time unit tests are broken if CFE_PLATFORM_TIME_CFG_SRC_MET is set to true

aliounis, 2023-09-01:
**Describe the bug**
When attempting to run unit/coverage tests (SIMULATION=NATIVE, ENABLE_UNIT_TESTS=true) there is a compilation error for the time_UT.c unit test file if the CFE_PLATFORM_TIME_CFG_SRC_MET flag is set to true for the project.  The compilation error appears to causes by references to undefined members of the CFE_TIME_Global structure (at minimum these are references to `ClockSetState`, `AtToneMET`, `AtToneDelay`, `AtToneLatch`, and `AtToneMET`.  After doing a cursory investigation, it looks like these may have been moved to a substructure array `ReferenceState`, though I'm not certain since the unit test seems to imply they were scalar values but `ReferenceState` is an array of structs.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a fresh cFS build
2. Change the CFE_PLATFORM_TIME_CFG_SRC_MET to true
3. Configure with SIMULATION=native, ENABLE_UNIT_TESTS=true
4. Attempt to build

**Expected behavior**
the unit tests should build and pass

**Code snips**
If applicable, add references to the software.

**System observed on:**
 - Hardware Intel Xeon
 - OS: ubuntu 22.4
 - Versions at least >= 7.0.0-rc4 (confirmed the issue is still present in main)

**Additional context**
N/A

**Reporter Info**
Andrew Liounis, NASA/GSFC

aliounis, 2023-09-01:
It looks like htis also might be an issue for other TIME_CFG_SRC selections

thnkslprpt, 2024-09-28:
@skliper already applied a fix for `CFE_PLATFORM_TIME_CFG_SRC_TIME` [here](https://github.com/nasa/cFE/pull/2099) and that options builds fine (and the tests all pass).

I'll put a quick PR in for the other 2 options.