nasa/cFE #2460: Performance Improvements for SB Transmit and Receive APIs

dzbaker, 2023-10-26:
**Is your feature request related to a problem? Please describe.**
The SB Transmit and Receive APIs tend to be the biggest performance bottleneck on a mission.

**Describe the solution you'd like**
Consider restoring the previous locking strategy.

**Additional context**
From RST:
SB Transmit and Receive APIs are generally the largest performance bottlenecks on mission. Changes in cFS since previous flight heritage missions have significantly decreased performance due to additional SB Lock calls (implemented to improve the modularity of the code, but without consideration for performance impacts).

Restoring the earlier Locking strategy reduces overall Spacecraft C&DH FSW utilization by 8.7% under nominal conditions, but could potentially be even greater once all subsystems are present.

**Requester Info**
Kequan Luu/NASA GSFC Code 582

dzbaker, 2023-10-26:
CCB 26 October 2023: Corwin will reach out to Kequan to identify which change specifically this is referring to.

dmknutsen, 2023-10-26:
@dzbaker Kequan may be referring to Fix #2381 

It was not merged due to potential downstream impacts on performance.  We are still setting up the lab to better characterize performance impacts on flight like targets.  There are multiple pull requests on hold until further performance characterization can be completed.

avan989, 2024-10-20:
@chillfig @dzbaker @dmknutsen https://github.com/nasa/cFE/pull/2381 was merged. We need to close the ticket.