nasa/cFE #2574: CFE_SB_ReceiveBuffer times out immediately if the system time changes on POSIX

ahatstat, 2024-07-03:
**Describe the bug**
When pending on `CFE_SB_ReceiveBuffer` with a timeout on a POSIX system, a change to the system time causes the function to immediately return with a timeout error code or pend longer than specified depending on the direction of the time change.

**To Reproduce**
`CFE_SB_ReceiveBuffer` with a timeout of 10 seconds on a Linux system.  Change the system time ahead by 1 hour.  The function will return immediately with code CFE_SB_TIME_OUT regardless of the actual elapsed time.

**Expected behavior**
The function should always wait for the specified timeout regardless of any change to the system time.  If this is not practical, it would be helpful to document this behavior in the function doc string. 

**System observed on:**
ARM Cortex A9
Linux
Versions Draco-RC5

**Additional context**
The implementation calls OSAL function `OS_QueueGet`.  The POSIX implementation calls `mq_timedreceive` which is known to [suffer from this problem](https://e2e.ti.com/support/processors-group/processors/f/processors-forum/548380/mq_timedreceive-and-changing-the-real-time-clock). 

**Reporter Info**
Andrew Hatstat, Geost

irowebbn, 2024-09-11:
Can confirm, I have experienced this same problem and came to the same conclusion as you. The mq API is simply not sufficient for working with multiple clocks.

dzbaker, 2024-12-19:
CCB 19 December 2024: Will address through documentation.