nasa/cFE #2604: Project import using Make fails prematurely

GallopingTurtle, 2024-09-23:
**Describe the bug**
When cloning cFS and after configuring the Makefile/CMake files, CLion is unable to properly import the project.

Clion runs the command `make --just-print --print-directory --keep-going all` which typically errors out, but now does so before anything useful happens.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone cFS on or after `f378320af1d9a1b2399dabfde5f6b22af3b1416b`
2. Copy `cfe/cmake/Makefile.sample` to the project root as well as `cfe/cmake/sample_defs`
3. run `make prep` then `make --just-print --print-directory --keep-going all`
4. Observe how quickly the makefile errors out
5. Switch to a commit before `f378320af1d9a1b2399dabfde5f6b22af3b1416b`
6. `make distclean && make prep` and `make --just-print --print-directory --keep-going all`
7. Observe how much more information is provided (this is sufficient for CLion to detect the project correctly)

**Expected behavior**
At the end of this import process, CLion should be able to provide code sense and other features.

**System observed on:**
 - OS: Ubuntu 22.04
 - cFS hash: `f378320af1d9a1b2399dabfde5f6b22af3b1416b` (last known working hash: `0befc9779e348b8af9d403a7f6d858a5474bc907`)
 - cFE hash: `dab2f42e9d54b187e6d18140d7c961cd88c2b6d1` (last known working hash: `b524c691ecfde656b29b513c84be9aaafc96c453`)

It would seem that the offending commit stems from #2587

**Additional context**
The logs provided by CLion are as follows:
```
make --just-print --print-directory --keep-going all
make: Entering directory 'cFS'
/usr/bin/make --no-print-directory -C "build" mission-all
/usr/bin/cmake -ScFS/cfe -BcFS/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/make -s -f CMakeFiles/Makefile2 mission-all
/usr/bin/cmake -ScFS/cfe -BcFS/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start cFS/build/CMakeFiles 9
/usr/bin/make -s -f CMakeFiles/Makefile2 CMakeFiles/mission-all.dir/all
/usr/bin/make -s -f tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/build.make tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/depend
cd cFS/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" cFS/cfe cFS/tools/elf2cfetbl cFS/build cFS/build/tools/elf2cfetbl cFS/build/tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/DependInfo.cmake --color=
/usr/bin/make -s -f tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/build.make tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/build
/usr/bin/cmake -E cmake_echo_color --switch= --green --progress-dir=cFS/build/CMakeFiles --progress-num=13 "Building C object tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/elf2cfetbl.c.o"
cd cFS/build/tools/elf2cfetbl && /usr/bin/cc -D_XOPEN_SOURCE=600 -IcFS/build/inc -IcFS/cfe/modules/core_api/fsw/inc -IcFS/osal/src/os/inc -IcFS/psp/fsw/inc -g -std=c99 -pedantic -Wall -Wstrict-prototypes -Wwrite-strings -Wpointer-arith -Wcast-align -Werror -MD -MT tools/elf2cfetbl/CMakeFiles/elf2cfetbl.dir/elf2cfetbl.c.o -MF CMakeFiles/elf2cfetbl.dir/elf2cfetbl.c.o.d -o CMakeFiles/elf2cfetbl.dir/elf2cfetbl.c.o -c cFS/tools/elf2cfetbl/elf2cfetbl.c
/usr/bin/cmake -E cmake_echo_color --switch= --green --bold --progress-dir=cFS/build/CMakeFiles --progress-num=14 "Linking C executable elf2cfetbl"
cd cFS/build/tools/elf2cfetbl && /usr/bin/cmake -E cmake_link_script CMakeFiles/elf2cfetbl.dir/link.txt --verbose=
/usr/bin/cmake -E cmake_echo_color --switch= --progress-dir=cFS/build/CMakeFiles --progress-num=13,14 "Built target elf2cfetbl"
/usr/bin/make -s -f CMakeFiles/cfe-build-env.dir/build.make CMakeFiles/cfe-build-env.dir/depend
cd cFS/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" cFS/cfe cFS/cfe cFS/build cFS/build cFS/build/CMakeFiles/cfe-build-env.dir/DependInfo.cmake --color=
/usr/bin/make -s -f CMakeFiles/cfe-build-env.dir/build.make CMakeFiles/cfe-build-env.dir/build
cd cFS/cfe && /usr/bin/cmake -D BIN=cFS/build -P cFS/cfe/cmake/generate_build_env.cmake
/usr/bin/cmake -E cmake_echo_color --switch= --progress-dir=cFS/build/CMakeFiles --progress-num= "Built target cfe-build-env"
/usr/bin/make -s -f CMakeFiles/cfe-module-version.dir/build.make CMakeFiles/cfe-module-version.dir/depend
cd cFS/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" cFS/cfe cFS/cfe cFS/build cFS/build cFS/build/CMakeFiles/cfe-module-version.dir/DependInfo.cmake --color=
/usr/bin/make -s -f CMakeFiles/cfe-module-version.dir/build.make CMakeFiles/cfe-module-version.dir/build
cd cFS/cfe && /usr/bin/cmake -D BIN=cFS/build -P cFS/cfe/cmake/generate_git_module_version.cmake
/usr/bin/cmake -E cmake_echo_color --switch= --progress-dir=cFS/build/CMakeFiles --progress-num= "Built target cfe-module-version"
/usr/bin/make -s -f CMakeFiles/mission-version.dir/build.make CMakeFiles/mission-version.dir/depend
cd cFS/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" cFS/cfe cFS/cfe cFS/build cFS/build cFS/build/CMakeFiles/mission-version.dir/DependInfo.cmake --color=
/usr/bin/make -s -f CMakeFiles/mission-version.dir/build.make CMakeFiles/mission-version.dir/build
/usr/bin/cmake -E cmake_echo_color --switch= --progress-dir=cFS/build/CMakeFiles --progress-num= "Built target mission-version"
/usr/bin/make -s -f cfeconfig_platformdata_tool/CMakeFiles/cfeconfig_platformdata_i686_linux_gnu_default_cpu1.dir/build.make cfeconfig_platformdata_tool/CMakeFiles/cfeconfig_platformdata_i686_linux_gnu_default_cpu1.dir/depend
cd cFS/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" cFS/cfe cFS/cfe/modules/config/tool cFS/build cFS/build/cfeconfig_platformdata_tool cFS/build/cfeconfig_platformdata_tool/CMakeFiles/cfeconfig_platformdata_i686_linux_gnu_default_cpu1.dir/DependInfo.cmake --color=
/usr/bin/make -s -f cfeconfig_platformdata_tool/CMakeFiles/cfeconfig_platformdata_i686_linux_gnu_default_cpu1.dir/build.make cfeconfig_platformdata_tool/CMakeFiles/cfeconfig_platformdata_i686_linux_gnu_default_cpu1.dir/build
make[4]: *** No rule to make target 'cfeconfig_platformdata_tool/CMakeFiles/cfeconfig_platformdata_i686_linux_gnu_default_cpu1.dir/cfeconfig_platformdata_table.c.o', needed by 'cfeconfig_platformdata_tool/cfeconfig_platformdata_tool'.
make[3]: *** [CMakeFiles/Makefile2:726: cfeconfig_platformdata_tool/CMakeFiles/cfeconfig_platformdata_tool.dir/all] Error 2
make[2]: *** [CMakeFiles/Makefile2:232: CMakeFiles/mission-all.dir/rule] Error 2
make[1]: *** [Makefile:179: mission-all] Error 2
make: *** [Makefile:119: all] Error 2
/usr/bin/cmake -E cmake_echo_color --switch= --green --progress-dir=cFS/build/CMakeFiles --progress-num=3 "Building C object cfeconfig_platformdata_tool/CMakeFiles/cfeconfig_platformdata_i686_linux_gnu_default_cpu1.dir/cfeconfig_platformdata_table.c.o"
cd cFS/build/cfeconfig_platformdata_tool && /usr/bin/cc -DMAP_ENTRY_LIST=CFECONFIG_PLATFORMDATA_i686_linux_gnu_default_cpu1 -D_XOPEN_SOURCE=600 -IcFS/build/inc -IcFS/cfe/modules/config/tool/../fsw/inc -IcFS/build/i686-linux-gnu/default_cpu1/inc -g -std=c99 -pedantic -Wall -Wstrict-prototypes -Wwrite-strings -Wpointer-arith -Wcast-align -Werror -MD -MT cfeconfig_platformdata_tool/CMakeFiles/cfeconfig_platformdata_i686_linux_gnu_default_cpu1.dir/cfeconfig_platformdata_table.c.o -MF CMakeFiles/cfeconfig_platformdata_i686_linux_gnu_default_cpu1.dir/cfeconfig_platformdata_table.c.o.d -o CMakeFiles/cfeconfig_platformdata_i686_linux_gnu_default_cpu1.dir/cfeconfig_platformdata_table.c.o -c cFS/cfe/modules/config/tool/cfeconfig_platformdata_table.c
/usr/bin/cmake -E cmake_echo_color --switch= --progress-dir=cFS/build/CMakeFiles --progress-num=3 "Built target cfeconfig_platformdata_i686_linux_gnu_default_cpu1"
/usr/bin/make -s -f cfeconfig_platformdata_tool/CMakeFiles/cfeconfig_platformdata_tool.dir/build.make cfeconfig_platformdata_tool/CMakeFiles/cfeconfig_platformdata_tool.dir/depend
[... 16 lines cut ...]
make install
```

still works as intended.

You will also notice between offending and non offending commits a block in the output of the make containing:

```
make -s -f apps/ci_lab/CMakeFiles/ci_lab.dir/build.make apps/ci_lab/CMakeFiles/ci_lab.dir/depend
cd cFS/build/i686-linux-gnu/default_cpu1 && /usr/bin/cmake -E cmake_depends "Unix Makefiles" cFS/cfe cFS/apps/ci_lab cFS/build/i686-linux-gnu/default_cpu1 cFS/build/i686-linux-gnu/default_cpu1/apps/ci_lab cFS/build/i686-linux-gnu/default_cpu1/apps/ci_lab/CMakeFiles/ci_lab.dir/DependInfo.cmake --color=
make -s -f apps/ci_lab/CMakeFiles/ci_lab.dir/build.make apps/ci_lab/CMakeFiles/ci_lab.dir/build
[... cut ...]
/usr/bin/cmake -E cmake_echo_color --switch= --green --bold --progress-dir=cFS/build/i686-linux-gnu/default_cpu1/CMakeFiles --progress-num= "Linking C shared module ci_lab.so"
cd cFS/build/i686-linux-gnu/default_cpu1/apps/ci_lab && /usr/bin/cmake -E cmake_link_script CMakeFiles/ci_lab.dir/link.txt --verbose=
/usr/bin/cmake -E cmake_echo_color --switch= --progress-dir=cFS/build/i686-linux-gnu/default_cpu1/CMakeFiles --progress-num=2 "Built target ci_lab"
```

which (including lines prior) gives CLion the information it needs to understand how `ci_lab` is built. Without it CLion is clueless.

jphickey, 2024-12-19:
Can CLion be configured to use `compile_commands.json` like VS Code?

I looked into this a bit -- the problem here relates to using `--just-print` ... this option is telling make not to actually execute the commands, just print them.  But in this case the file `cfeconfig_platformdata_table.c` is a generated file.  The dependency is in there, but because its not running any commands, the file isn't built, and thus doesn't exist when make gets to that point.

I think the option `--just-print` is fundamentally incompatible with this type of makefile construct, where the output of one command is used as an input to a sub-make command.  (and this is not an unusual thing to do, by any stretch).

At any rate, I don't see this as a flaw with the CFE makefiles, I see it as a flaw with the assumptions of the `--just-print` option.  Naturally, if it just prints a command and doesn't run the command, the expected outputs aren't going to exist, and thus the next rule that depends on that output file won't find it.

The `compile_commands.json` file is intended to address this problem for IDEs and it is the intended way of informing an IDE of what the compiler commands are supposed to be.

jphickey, 2024-12-19:
I found this link, it seems it does support the JSON file method: https://www.jetbrains.com/help/clion/compilation-database.html

jphickey, 2024-12-19:
Please  note that CFE/CFS is not just a simple CMake project.  This is necessary to produce binaries and table files for multiple different architectures with a single "make" and build tree.  However this layering means that it doesn't match what most IDEs expect when you indicate something as a "CMake project".    For example with VS Code and Eclipse - I configure the IDE as a general C/C++ project with custom makefiles.   I do _not_ use whatever pre-canned logic it as for a CMake project, as CFE/CFS is more complicated than your average CMake project.

dzbaker, 2025-01-02:
CCB 2 January 2025: Closing as not an issue that needs to be fixed. "--just-print" is not expected to work as it does not produce dependent files.

closed 2025-01-02 by dzbaker