nasa/cFE #2627: TBL verify check for CFE_PLATFORM_TBL_MAX_SNGL_TABLE_SIZE does not match documented description

irowebbn, 2025-01-14:
**Describe the bug**

According to the desciption in the platform config header, the value for `CFE_PLATFORM_TBL_MAX_SNGL_TABLE_SIZE)` must be "small enough to allow for `#CFE_PLATFORM_TBL_MAX_SIMULTANEOUS_LOADS` number of tables to fit into `#CFE_PLATFORM_TBL_BUF_MEMORY_BYTES`." (see https://github.com/nasa/cFE/blob/d6e610756cc8f2caddf465a92dff92fcd26d8e33/cmake/sample_defs/example_platform_cfg.h#L1369).

However, in `modules/tbl/fsw/src/cfe_tbl_verify.h` , the logic for checking this appears to be off by one:

```c
#if ((CFE_PLATFORM_TBL_MAX_SIMULTANEOUS_LOADS + 1) * CFE_PLATFORM_TBL_MAX_SNGL_TABLE_SIZE) > \
    CFE_PLATFORM_TBL_BUF_MEMORY_BYTES
#error Shared buffers and table of size CFE_PLATFORM_TBL_MAX_SNGL_TABLE_SIZE cannot be greater than memory pool size of CFE_PLATFORM_TBL_BUF_MEMORY_BYTES!
#endif

```

This leads to compliant configuration values failing to compile.

**To Reproduce**
Steps to reproduce the behavior:
1. Edit `CFE_PLATFORM_TBL_MAX_SNGL_TABLE_SIZE` and ` CFE_PLATFORM_ES_MAX_BLOCK_SIZE` to be a large value, such as `196608`
2. Edit `CFE_PLATFORM_TBL_BUF_MEMORY_BYTES` to be equal to `CFE_PLATFORM_TBL_MAX_SNGL_TABLE_SIZE * CFE_PLATFORM_TBL_MAX_SIMULTANEOUS_LOADS` (in this case 786432, for the default simultaneous load value of 4).
3. Try to compile
4. See error

```
[15%] Building C object tbl/CMakeFiles/tbl.dir/fsw/src/cfe_tbl_task.c.o
In file included from cfe/modules/tbl/fsw/src/cfe_tbl_task.c:35:
cfe/modules/tbl/fsw/src/cfe_tbl_verify.h:39:2: error: #error Shared buffers and table of size CFE_PLATFORM_TBL_MAX_SNGL_TABLE_SIZE cannot be greater than memory pool size of CFE_PLATFORM_TBL_BUF_MEMORY_BYTES!
```
**Expected behavior**
Successful compilation

**System observed on:**
 - Hardware: x86_64
 - OS: Rocky Linux 8
 - Versions `draco-rc4` tag

**Reporter Info**
Isaac Rowe, NASA JSC/Jacobs Technology

jkeo180, 2026-01-05:
may i work on this issue?

irowebbn, 2026-01-05:
@jkeo180 I have a fix at #2632, but the repo maintainers have not responded.