nasa/cFE #2663: Unregistered AppId error with CFE_EVS_SendEventWithAppID() during cFE task startup

myc-yang, 2025-08-06:
**Describe the bug**
When the ES and the EVS tasks initializes, they make various CFE_SB API calls (i.e. `CFE_SB_CreatePipe()` and `CFE_SB_Subscribe()`) which in turn call `CFE_EVS_SendEventWithAppID()` passing in the AppId for the CFE_SB.  `CFE_EVS_SendEventWithAppID()` then returns a `CFE_EVS_APP_ILLEGAL_APP_ID` presumably because CFE_SB has not been registered with event services.  This might have gone unnoticed because there is no event outputted nor is the return code set to any variable.  

**To Reproduce**
Steps to reproduce the behavior:
1. make SIMULATION=native prep
2. make
3. make install
4. Place a breakpoint here - https://github.com/nasa/cFE/blob/main/modules/evs/fsw/src/cfe_evs.c#L185
5. Run the gdb and see it stopped at the breakpoint.  

**System observed on:**
 - OS: Ubuntu 22.04.5 LTS
 - cFE v6.7

**Reporter Info**
Michael Yang - NASA/GSFC

dzbaker, 2025-09-04:
CCB 4 September 2025: Tracked internally. Development team will work on this ticket.

philphauler, 2026-09-08:
Hi all, I opened #2820 for this and after a closer read I am closing it: the change there is vacuous (EVS_GetAppDataByID is deterministic in the ID, so a caller whose AppId equals the event's AppId already resolves to the same record; the UT only passed through a deferred stub return code). Sharing what the read turned up in case it helps whoever picks this up.

The ILLEGAL_APP_ID at startup comes from SB, not from the caller EVS is checking. ES and EVS each create their command pipe inside their own task init (cfe_es_task.c:367, cfe_evs_task.c:289). SB attributes pipe-creation events to itself using CFE_SB_Global.AppId, but that field is only filled in CFE_SB_AppInit (cfe_sb_task.c:138) when the SB task runs. Whichever core app's init runs before that sees SB call CFE_EVS_SendEventWithAppID with an undefined AppId, which is exactly the error reported here.

Two ways to fix that I can see: have SB resolve its AppId lazily on first use (a one-line check in the event path), or have CFE_SB_EarlyInit record it since ES already knows the SB app id at that point. Happy to write either one with a UT if the team has a preference; otherwise I will leave this with you since it was marked as tracked internally. Have a great day!